ARMmbed/mbed-os-example-filesystem #15: NRF82840DK External Flash Hangs

kegilbert, 2017-12-15:
Ran with pins:
```
 "spif-driver-spi-mosi": {
     "value": "p20"
 },
 "spif-driver-spi-miso": {
     "value": "p21"
 },
 "spif-driver-spi-clk": {
     "value": "p19"
 },
 "spif-driver-spi-cs": {
     "value": "p17"
```
Source: http://infocenter.nordicsemi.com/index.jsp?topic=%2Fcom.nordic.infocenter.nrf52%2Fdita%2Fnrf52%2Fdevelopment%2Fnrf52840_pdk%2Fexternal_memory.html

Additionally tried with SPI0 from the pinnames.h file:
```
 SPI_PSELMOSI0 = P1_13, 
 SPI_PSELMISO0 = P1_14, 
 SPI_PSELSS0   = P1_12, 
 SPI_PSELSCK0  = P1_15, 
```

___________________________________________________________________________-
With the first set of defined pins from the product documentation I get the the following output followed by the hang:
```
--- Mbed OS filesystem example ---
Mounting the filesystem... Fail :(
No filesystem found, formatting...
```

@geky 
@senthilr

geky, 2017-12-15:
Ah! There is an outstanding issue with wait calls on the NRF52840. @dlfryar ran into this issue earlier.

I believe this pr fixes the problem: https://github.com/ARMmbed/mbed-os/pull/5595

Does the example work with https://github.com/ARMmbed/mbed-os/pull/5595?

dlfryar, 2017-12-15:
You probably also need https://github.com/ARMmbed/mbed-os/pull/5606.  There is still an outstanding problem with 5595 and assertions, but the example should run then assert.

geky, 2018-03-09:
Double checked by @kegilbert, this has been resolved by https://github.com/ARMmbed/mbed-os/pull/5606

Feel free to reopen if that's not the case

closed 2018-03-09 by geky