ARMmbed/mbed-os-example-filesystem #16: numbers.txt incrememts imporoperly with DataFlashBlockDevice on K64F

cmonr, 2017-12-15:
Output:

```
--- Mbed OS filesystem example ---
Mounting the filesystem... lfs error: Corrupted dir pair at 0 1
lfs error: Invalid superblock at 72813 16842752
Fail :(
No filesystem found, formatting... OK
Opening "/fs/numbers.txt"... Fail :(
No file found, creating a new file... OK
Writing numbers (10/10)... OK
Seeking file... OK
Incrementing numbers (10/10)... OK
Closing "/fs/numbers.txt"... OK
Opening the root directory... OK
root directory:
    .
    ..
    numbers.txt
Closing the root directory... OK
Opening "/fs/numbers.txt"... OK
numbers:
    1
    1
    2
    3
    4
    5
    6
    7
    8
    9
Closing "/fs/numbers.txt"... OK
Unmounting... OK
Mbed OS filesystem example done!
--- Mbed OS filesystem example ---
Mounting the filesystem... OK
Opening "/fs/numbers.txt"... OK
Incrementing numbers (10/10)... OK
Closing "/fs/numbers.txt"... OK
Opening the root directory... OK
root directory:
    .
    ..
    numbers.txt
Closing the root directory... OK
Opening "/fs/numbers.txt"... OK
numbers:
    2
    1
    2
    3
    4
    5
    6
    7
    8
    9
Closing "/fs/numbers.txt"... OK
Unmounting... OK
Mbed OS filesystem example done!
```

The following code was added into main.cpp to test:
```
DigitalOut a(D10, 1);
DigitalOut b(D9, 1);
DigitalOut c(D8, 1);
DigitalOut d(D7, 1);
DigitalOut e(D6, 1);
DigitalOut f(D5, 1);
DigitalOut g(D4, 1);

DataFlashBlockDevice bd(D11, D12, D13, D8);

// File system declaration
LittleFileSystem fs("fs");
//FATFileSystem fs("fs");
```

kegilbert, 2017-12-15:
We've determined it only happens on ARMC5.

cmonr, 2017-12-18:
Fyi, also affects UBLOX_EVK_ODIN_W2

geky, 2017-12-19:
More investigation: This only happens on ARMCC, but is at least consistent across all of the block devices and file systems. This must be a stdlib or retargetting issue, probably related to ARMCC's seek function.

Note: This issue was already present in Mbed OS 5.6.

Requires more investigation

cmonr, 2017-12-19:
@geky Is there already an issue against mbed-os 5.6 that this can be referenced to, or should a new issue be made in that repo?

geky, 2017-12-19:
Oh probably. I'm consider issues on this repo as important as issues on mbed OS, but you're right that's where the problem is.

Are you able to raise an issue there?

cmonr, 2017-12-19:
Anyone can raise an issue there!
https://github.com/ARMmbed/mbed-os/issues

When you mentioned that the issue was already present in mbed-os 5.6, were you referring to the seek issue, or something else?

cmonr, 2017-12-19:
In general, I don't think it's wise to raise an issue unless we're able to add specific details on how to reliably reproduce the issue.

If there were one already made, I would have simply referenced the issues to each other for tracking purposes.

geky, 2017-12-19:
Ah, I meant the issue was not introduced by 5.7

kegilbert, 2018-03-05:
Reproduced during 5.8 OOBing on LittleFS and FATFS for ARMC5.

@geky @cmonr

adbridge, 2018-03-09:
@geky So this can be closed once #34 is merged ?

fredlee12001, 2018-03-19:
Retested with RC2 release (c76e4f121387fcff7cac233fbb0137c6e0bc6ba4).  The issue was still there.

cmonr, 2018-03-19:
@fredlee12001 Can you provide steps to reproduce the issue?

fredlee12001, 2018-03-20:
@cmonr ，  I'm testing with below commands:
`mbed import mbed-os-example-filesystem`
 `cd mbed-os-example-filesystem`
 `mbed update mbed-os-5.8.0-oob`
 `mbed compile -t ARM -m K64F -c -f`

Then reset the K64F, we could see prints as below:

> Mounting the filesystem... lfs error:469: Corrupted dir pair at 0 1
lfs error:2202: Invalid superblock at 64645 500
Fail :(
No filesystem found, formatting... OK
Opening "/fs/numbers.txt"... Fail :(
No file found, creating a new file... OK
Writing numbers (0/10)...
Writing numbers (1/10)...
Writing numbers (2/10)...
Writing numbers (3/10)...
Writing numbers (4/10)...
Writing numbers (5/10)...
Writing numbers (6/10)...
Writing numbers (7/10)...
Writing numbers (8/10)...
Writing numbers (9/10)...
Writing numbers (10/10)... OK
Seeking file... OK
Incrementing numbers (0/10)...
Incrementing numbers (1/10)...
Incrementing numbers (2/10)...
Incrementing numbers (3/10)...
Incrementing numbers (4/10)...
Incrementing numbers (5/10)...
Incrementing numbers (6/10)...
Incrementing numbers (7/10)...
Incrementing numbers (8/10)...
Incrementing numbers (9/10)...
Incrementing numbers (10/10)... OK
Closing "/fs/numbers.txt"... OK
Opening the root directory... OK
root directory:
    .
    ..
    numbers.txt
Closing the root directory... OK
Opening "/fs/numbers.txt"... OK
numbers:
    1
    1
    2
    3
    4
    5
    6
    7
    8
    9
ÿ
Closing "/fs/numbers.txt"... OK
Unmounting... OK
Mbed OS filesystem example done!

kegilbert, 2018-03-20:
The fix here:
https://github.com/ARMmbed/mbed-os-example-filesystem/commit/29a91bfe10af6c5fa8e879b85df6fca29029ca31
is only on master it looks like, it needs to be cherry picked to the release branch.

dannybenor, 2018-06-10:
Already on master

closed 2018-06-10 by dannybenor