ARMmbed/mbed-os-example-filesystem #18: FAT is too... chunky?... for Data Flash

cmonr, 2017-12-15:
```
--- Mbed OS filesystem example ---
Mounting the filesystem... mbed assertation failed: ssize >= FF_MIN_SS && ssize <= FF_MAX_SS, file: .\mbed-os\features\filesystem\fat\FATFileSystem.cpp, l8▒|

```

Compiler: GCC_ARM
Target: K64F

Used with Mbed Flash Memory Shield

Data flash initialization code:
```
DigitalOut a(D4, 1);
DigitalOut b(D5, 1);
DigitalOut c(D6, 1);
DigitalOut d(D7, 1);
DigitalOut e(D8, 1);
DigitalOut f(D9, 1);
DigitalOut g(D10, 1);
		
DataFlashBlockDevice bd(D11, D12, D13, D8);
```

geky, 2017-12-18:
Good find 👍, the minimum block count of FAT has popped up quite a few times before.

We should add a note in the README of the limitations for the available filesystems.

geky, 2017-12-20:
We've sinced added this note: https://github.com/ARMmbed/mbed-os-example-filesystem/pull/26/files

Feel free to reopen this issue if you think it needs more work

closed 2017-12-20 by geky