ARMmbed/mbed-os-example-filesystem #20: ODIN-W2 stuck at FS mount w/ ARM compiler

cmonr, 2017-12-18:
Note: HW config needed. UART3 is used for serial in lieu of UART1 to test with both AT45 parts. [Reference](https://os.mbed.com/platforms/ublox-EVK-ODIN-W2/).

Output:
```
--- Mbed OS filesystem example ---
Mounting the filesystem... 
```

Diff:
```
diff --git a/main.cpp b/main.cpp
index bb30a33..01e499c 100644
--- a/main.cpp
+++ b/main.cpp
@@ -16,6 +16,7 @@
 #include "mbed.h"
 #include <stdio.h>
 #include <errno.h>
+#include <serial.h>
 
 // Block devices
 #include "SPIFBlockDevice.h"
@@ -27,41 +28,38 @@
 #include "LittleFileSystem.h"
 #include "FATFileSystem.h"
 
+Serial ser3(D1, D0);
 
 // Physical block device, can be any device that supports the BlockDevice API
-SPIFBlockDevice bd(
-        MBED_CONF_SPIF_DRIVER_SPI_MOSI,
-        MBED_CONF_SPIF_DRIVER_SPI_MISO,
-        MBED_CONF_SPIF_DRIVER_SPI_CLK,
-        MBED_CONF_SPIF_DRIVER_SPI_CS);
+DataFlashBlockDevice bd(D11, D12, D13, D7);
 
 // File system declaration
-LittleFileSystem fs("fs");
-
+//LittleFileSystem fs("fs");
+FATFileSystem fs("fs");
 
 // Set up the button to trigger an erase
 InterruptIn irq(BUTTON1);
 void erase() {
-    printf("Initializing the block device... ");
+    ser3.printf("Initializing the block device... ");
     fflush(stdout);
     int err = bd.init();
-    printf("%s\n", (err ? "Fail :(" : "OK"));
+    ser3.printf("%s\n", (err ? "Fail :(" : "OK"));
     if (err) {
         error("error: %s (%d)\n", strerror(-err), err);
     }
 
-    printf("Erasing the block device... ");
+    ser3.printf("Erasing the block device... ");
     fflush(stdout);
     err = bd.erase(0, bd.size());
-    printf("%s\n", (err ? "Fail :(" : "OK"));
+    ser3.printf("%s\n", (err ? "Fail :(" : "OK"));
     if (err) {
         error("error: %s (%d)\n", strerror(-err), err);
     }
 
-    printf("Deinitializing the block device... ");
+    ser3.printf("Deinitializing the block device... ");
     fflush(stdout);
     err = bd.deinit();
-    printf("%s\n", (err ? "Fail :(" : "OK"));
+    ser3.printf("%s\n", (err ? "Fail :(" : "OK"));
     if (err) {
         error("error: %s (%d)\n", strerror(-err), err);
     }
@@ -70,58 +68,58 @@ void erase() {
 
 // Entry point for the example
 int main() {
-    printf("--- Mbed OS filesystem example ---\n");
+    ser3.printf("--- Mbed OS filesystem example ---\n");
 
     // Setup the irq in case we want to use it
     irq.fall(erase);
 
     // Try to mount the filesystem
-    printf("Mounting the filesystem... ");
+    ser3.printf("Mounting the filesystem... ");
     fflush(stdout);
     int err = fs.mount(&bd);
-    printf("%s\n", (err ? "Fail :(" : "OK"));
+    ser3.printf("%s\n", (err ? "Fail :(" : "OK"));
     if (err) {
         // Reformat if we can't mount the filesystem
         // this should only happen on the first boot
-        printf("No filesystem found, formatting... ");
+        ser3.printf("No filesystem found, formatting... ");
         fflush(stdout);
         err = fs.reformat(&bd);
-        printf("%s\n", (err ? "Fail :(" : "OK"));
+        ser3.printf("%s\n", (err ? "Fail :(" : "OK"));
         if (err) {
             error("error: %s (%d)\n", strerror(-err), err);
         }
     }
 
     // Open the numbers file
-    printf("Opening \"/fs/numbers.txt\"... ");
+    ser3.printf("Opening \"/fs/numbers.txt\"... ");
     fflush(stdout);
     FILE *f = fopen("/fs/numbers.txt", "r+");
-    printf("%s\n", (!f ? "Fail :(" : "OK"));
+    ser3.printf("%s\n", (!f ? "Fail :(" : "OK"));
     if (!f) {
         // Create the numbers file if it doesn't exist
-        printf("No file found, creating a new file... ");
+        ser3.printf("No file found, creating a new file... ");
         fflush(stdout);
         f = fopen("/fs/numbers.txt", "w+");
-        printf("%s\n", (!f ? "Fail :(" : "OK"));
+        ser3.printf("%s\n", (!f ? "Fail :(" : "OK"));
         if (!f) {
             error("error: %s (%d)\n", strerror(errno), -errno);
         }
 
         for (int i = 0; i < 10; i++) {
-            printf("\rWriting numbers (%d/%d)... ", i, 10);
+            ser3.printf("\rWriting numbers (%d/%d)... ", i, 10);
             fflush(stdout);
             err = fprintf(f, "    %d\n", i);
             if (err < 0) {
-                printf("Fail :(\n");
+                ser3.printf("Fail :(\n");
                 error("error: %s (%d)\n", strerror(errno), -errno);
             }
         }
-        printf("\rWriting numbers (%d/%d)... OK\n", 10, 10);
+        ser3.printf("\rWriting numbers (%d/%d)... OK\n", 10, 10);
 
-        printf("Seeking file... ");
+        ser3.printf("Seeking file... ");
         fflush(stdout);
         err = fseek(f, 0, SEEK_SET);
-        printf("%s\n", (err < 0 ? "Fail :(" : "OK"));
+        ser3.printf("%s\n", (err < 0 ? "Fail :(" : "OK"));
         if (err < 0) {
             error("error: %s (%d)\n", strerror(errno), -errno);
         }
@@ -129,7 +127,7 @@ int main() {
 
     // Go through and increment the numbers
     for (int i = 0; i < 10; i++) {
-        printf("\rIncrementing numbers (%d/%d)... ", i, 10);
+        ser3.printf("\rIncrementing numbers (%d/%d)... ", i, 10);
         fflush(stdout);
 
         // Get current stream position
@@ -146,76 +144,76 @@ int main() {
         // Store number
         fprintf(f, "    %d\n", number);
     }
-    printf("\rIncrementing numbers (%d/%d)... OK\n", 10, 10);
+    ser3.printf("\rIncrementing numbers (%d/%d)... OK\n", 10, 10);
 
     // Close the file which also flushes any cached writes
-    printf("Closing \"/fs/numbers.txt\"... ");
+    ser3.printf("Closing \"/fs/numbers.txt\"... ");
     fflush(stdout);
     err = fclose(f);
-    printf("%s\n", (err < 0 ? "Fail :(" : "OK"));
+    ser3.printf("%s\n", (err < 0 ? "Fail :(" : "OK"));
     if (err < 0) {
         error("error: %s (%d)\n", strerror(errno), -errno);
     }
     
     // Display the root directory
-    printf("Opening the root directory... ");
+    ser3.printf("Opening the root directory... ");
     fflush(stdout);
     DIR *d = opendir("/fs/");
-    printf("%s\n", (!d ? "Fail :(" : "OK"));
+    ser3.printf("%s\n", (!d ? "Fail :(" : "OK"));
     if (!d) {
         error("error: %s (%d)\n", strerror(errno), -errno);
     }
 
-    printf("root directory:\n");
+    ser3.printf("root directory:\n");
     while (true) {
         struct dirent *e = readdir(d);
         if (!e) {
             break;
         }
 
-        printf("    %s\n", e->d_name);
+        ser3.printf("    %s\n", e->d_name);
     }
 
-    printf("Closing the root directory... ");
+    ser3.printf("Closing the root directory... ");
     fflush(stdout);
     err = closedir(d);
-    printf("%s\n", (err < 0 ? "Fail :(" : "OK"));
+    ser3.printf("%s\n", (err < 0 ? "Fail :(" : "OK"));
     if (err < 0) {
         error("error: %s (%d)\n", strerror(errno), -errno);
     }
 
     // Display the numbers file
-    printf("Opening \"/fs/numbers.txt\"... ");
+    ser3.printf("Opening \"/fs/numbers.txt\"... ");
     fflush(stdout);
     f = fopen("/fs/numbers.txt", "r");
-    printf("%s\n", (!f ? "Fail :(" : "OK"));
+    ser3.printf("%s\n", (!f ? "Fail :(" : "OK"));
     if (!f) {
         error("error: %s (%d)\n", strerror(errno), -errno);
     }
 
-    printf("numbers:\n");
+    ser3.printf("numbers:\n");
     while (!feof(f)) {
         int c = fgetc(f);
-        printf("%c", c);
+        ser3.printf("%c", c);
     }
 
-    printf("\rClosing \"/fs/numbers.txt\"... ");
+    ser3.printf("\rClosing \"/fs/numbers.txt\"... ");
     fflush(stdout);
     err = fclose(f);
-    printf("%s\n", (err < 0 ? "Fail :(" : "OK"));
+    ser3.printf("%s\n", (err < 0 ? "Fail :(" : "OK"));
     if (err < 0) {
         error("error: %s (%d)\n", strerror(errno), -errno);
     }
 
     // Tidy up
-    printf("Unmounting... ");
+    ser3.printf("Unmounting... ");
     fflush(stdout);
     err = fs.unmount();
-    printf("%s\n", (err < 0 ? "Fail :(" : "OK"));
+    ser3.printf("%s\n", (err < 0 ? "Fail :(" : "OK"));
     if (err < 0) {
         error("error: %s (%d)\n", strerror(-err), err);
     }
         
-    printf("Mbed OS filesystem example done!\n");
+    ser3.printf("Mbed OS filesystem example done!\n");
 }
 ```

cmonr, 2017-12-18:
Adding software pullups (`DigitalOut a(Dx, 1);`) does not change results.

geky, 2017-12-18:
Hmm, UART3 is on D7 and D6 right? Wouldn't that cause those chips to mess with the SPI bus?

cmonr, 2017-12-18:
UART3 is actually on D1 and D0. The flow control pins are on D7 and D6.

I _think_ D7 and D6 should be free since flow control is not being used.

kegilbert, 2018-03-05:
Is this on SPIF only? Works on SD Card (but still run into https://github.com/ARMmbed/mbed-os-example-filesystem/issues/16)

@cmonr

closed 2018-03-09 by geky