ARMmbed/mbed-os-example-filesystem #21: LittleFS failed for first couple of resets, then works?

cmonr, 2017-12-18:
Not sure if this is an actual issue, or if LittleFS is working as expected on failing test hardware.

Target: Nucleo F746ZG
Toolchain: GCC_ARM
Additional HW: Mbed Flash Memory Shield

Diff:
```
diff --git a/main.cpp b/main.cpp
index bb30a33..dfb3e75 100644
--- a/main.cpp
+++ b/main.cpp
@@ -27,17 +27,12 @@
 #include "LittleFileSystem.h"
 #include "FATFileSystem.h"
 
-
 // Physical block device, can be any device that supports the BlockDevice API
-SPIFBlockDevice bd(
-        MBED_CONF_SPIF_DRIVER_SPI_MOSI,
-        MBED_CONF_SPIF_DRIVER_SPI_MISO,
-        MBED_CONF_SPIF_DRIVER_SPI_CLK,
-        MBED_CONF_SPIF_DRIVER_SPI_CS);
+DataFlashBlockDevice bd(D11, D12, D13, D7);
 
 // File system declaration
 LittleFileSystem fs("fs");
-
+//FATFileSystem fs("fs");
 
 // Set up the button to trigger an erase
 InterruptIn irq(BUTTON1);
```

Output:
```
--- Mbed OS filesystem example ---
Mounting the filesystem... lfs error: Corrupted dir pair at 0 1
lfs error: Invalid superblock at 536874384 536874084
Fail :(
No filesystem found, formatting... Fail :(
error:  (-4001)
--- Mbed OS filesystem example ---
Mounting the filesystem... OK
Opening "/fs/numbers.txt"... lfs error: Corrupted dir pair at 3 2
Fail :(
No file found, creating a new file... lfs error: Corrupted dir pair at 3 2
Fail :(
error:  (-52)
--- Mbed OS filesystem example ---
Mounting the filesystem... lfs error: Invalid superblock at 0 1
Fail :(
No filesystem found, formatting... lfs debug: Bad block at 0
lfs warn: Superblock 0 has become unwritable
lfs error: Corrupted dir pair at 0 1
Fail :(
error:  (-52)
--- Mbed OS filesystem example ---
Mounting the filesystem... OK
Opening "/fs/numbers.txt"... OK
Incrementing numbers (10/10)... OK
Closing "/fs/numbers.txt"... OK
Opening the root directory... OK
root directory:
    .
    ..
    numbers.txt
Closing the root directory... OK
Opening "/fs/numbers.txt"... OK
numbers:
    5
    6
    7
    8
    9
    10
    11
    12
    13
    14
Closing "/fs/numbers.txt"... OK
Unmounting... OK
Mbed OS filesystem example done!
--- Mbed OS filesystem example ---
Mounting the filesystem... OK
Opening "/fs/numbers.txt"... OK
Incrementing numbers (10/10)... OK
Closing "/fs/numbers.txt"... OK
Opening the root directory... OK
root directory:
    .
    ..
    numbers.txt
Closing the root directory... OK
Opening "/fs/numbers.txt"... OK
numbers:
    6
    7
    8
    9
    10
    11
    12
    13
    14
    15
Closing "/fs/numbers.txt"... OK
Unmounting... OK
Mbed OS filesystem example done!
--- Mbed OS filesystem example ---
Mounting the filesystem... OK
Opening "/fs/numbers.txt"... OK
Incrementing numb▒--- Mbed OS filesystem example ---
Mounting the filesystem... OK
Opening "/fs/numbers.txt"... OK
Incrementing numbers (10/10)... OK
▒--- Mbed OS filesystem example ---
Mounting the filesystem... OK
Opening "/fs/numbers.txt"... OK
Incrementing numbers (10/10)... OK
Closing "/fs/numbers.txt"... OK
Opening the root directory... OK
root directory:
    .
    ..
    numbers.txt
Closing the root directory... OK
Opening "/fs/numbers.txt"... OK
numbers:
    7
    8
    9
    10
    11
    12
    13
    14
    15
    16
Closing "/fs/numbers.txt"... OK
Unmounting... OK
Mbed OS filesystem example done!
```

geky, 2017-12-18:
That's... weird. Also why did numbers start with 5? It must have been picking up an old filesystem?

This requires more investigation

cmonr, 2017-12-19:
It's definitely odd.

It's completely possible and likely that it picked up and continued from an older filesystem from a different board (one shield, many targets), but I don't think adding flash erasing to the testing would be a good idea.

geky, 2017-12-19:
The example is setup so if you push BUTTON1, it will erase the storage for you and give you a clean slate.

Unfortunately you just can't trust storage until it is formatted. Unless you are cryptographically signing the filesystem, a malicious filesystem could be constructed beforehand that looks fine but will break down later.

You can trust me though, I didn't do that to your board beforehand. 😛 

Most embedded systems come erased or at least usually don't look like a valid filesystem (complete with checksums), so it's not too concerning. Although if normally developer usage with randomly broken formats often leads to subtley broken filesystems it's worth looking into. I'd be curious if you could reproduce what you did after a full erase.

cmonr, 2017-12-19:
I'll give it another look tomorrow, and post what, if anything, I'm able to find.

cmonr, 2017-12-19:
Bummer. Wasn't able to reattempt the issue, since it looks like erasing the DataFlash block device wasn't working (reference new issue: https://github.com/ARMmbed/mbed-os-example-filesystem/issues/23). 

I'll switch to alternating FS to force the reformat, and see what comes up.

geky, 2018-03-09:
Related: https://github.com/ARMmbed/mbed-os/issues/5871

Since this is an enhancement, I'm going to close this for now. Feel free to continue in https://github.com/ARMmbed/mbed-os/issues/5871

closed 2018-03-09 by geky