ARMmbed/mbed-os-example-filesystem #23: Erase action in example does not erase DataFlash block device

cmonr, 2017-12-19:
Reference https://github.com/ARMmbed/mbed-os-example-filesystem/issues/22 for output and testing parameters.

Erasing and de-initializing the block device do not appear to work.

geky, 2017-12-20:
Oh huh... The erase function could be a no-op, so maybe we need to actually program 0xffs across the device... Or the dataflash block device could not be erasing correctly.

What happens? Does it print success and then still pick up the filesystem?

cmonr, 2017-12-20:
@geky Reference the link in the original comment for sample output.

kegilbert, 2018-03-09:
Reproduced it with IAR on an UBLOX EVK Odin W2 using the DataFlash shield. Everything else ran as expected.

davidsaada, 2018-06-11:
Example repo uses an old version of the DataFlash driver, which has a bug in the erase API (was fixed since). Will create a PR to use the up to date version.

davidsaada, 2018-06-12:
Example repo updated with fix.

closed 2018-06-12 by davidsaada

geky, 2018-06-12:
Thanks for taking care of this :+1: