ARMmbed/mbed-os-example-filesystem #24: FatFileSystem/HeapBlockDevice format returns -9

karsev, 2017-12-20:
### Description
- Type: Bug
- Priority: Critical

---------------------------------------------------------------
## Bug

**Target**
K64F

**Toolchain:**
GCC_ARM, ARM

**Toolchain version:**

**mbed-cli version:** 
1.1.1

**mbed-os sha:**
f4274b4 (HEAD, tag: mbed-os-5.7.0-rc3) Updating mbed-os to mbed-os-5.7.0-rc3

**DAPLink version:**

**Expected behavior**
- Using mbed-os-example-fat-filesystem
- Formatting and mounting works without errors.

**Actual behavior**
Formatting returns error code -9.
In the example heap block defined as -> HeapBlockDevice bd(128 * 512, 512);
if defined as -> HeapBlockDevice bd(1024 * 512, 512);
Starts working?
**Steps to reproduce**
run mbed-os-example-fat-filesystem example

geky, 2017-12-20:
Thanks for reporting the issue.

The FAT filesystem requires a minimum of 256 blocks. Unfortunately the spec doesn't allow the filesystem to be much smaller.

Though we do need to add a note about the minimum requirements for both of the filesystems.

Related https://github.com/ARMmbed/mbed-os-example-filesystem/issues/18

geky, 2017-12-20:
We've sinced added this note: https://github.com/ARMmbed/mbed-os-example-filesystem/pull/26/files

Feel free to reopen this issue if you think it needs more work

closed 2017-12-20 by geky