ARMmbed/mbed-os-example-filesystem #33: LittleFS - DataFlash on the Odin W2 [IAR] corrupts directory pairs

kegilbert, 2018-03-07:
Running the LittleFS example with the DataFlash shield on a Ublox EVK Odin W2 I get the following output after reseting the filesystem:

```
--- Mbed OS filesystem example ---
Mounting the filesystem... OK
Opening "/fs/numbers.txt"... OK
Incrementing numbers (10/10)... OK
Closing "/fs/numbers.txt"... OK
Opening the root directory... OK
root directory:
    .
    ..
    numbers.txt
Closing the root directory... OK
Opening "/fs/numbers.txt"... lfs error:469: Corrupted dir pair at 3 2
Fail :(
error: unknown error (-52)
--- Mbed OS filesystem example ---
Mounting the filesystem... OK
Opening "/fs/numbers.txt"... OK
Incrementing numbers (10/10)... OK
Closing "/fs/numbers.txt"... OK
Opening the root directory... OK
root directory:
    .
    ..
    numbers.txt
Closing the root directory... OK
Opening "/fs/numbers.txt"... OK
numbers:
    6
    7
    8
    9
    10
    11
    12
    13
    14
    15
7
2
Closing "/fs/numbers.txt"... OK
Unmounting... OK
Mbed OS filesystem example done!
```

__________________________________________________________
Attempt 2:

```
Initializing the block device... OK
Erasing the block device... OK
Deinitializing the block device... OK
--- Mbed OS filesystem example ---
Mounting the filesystem... OK
Opening "/fs/numbers.txt"... OK
Incrementing numbers (10/10)... OK
Closing "/fs/numbers.txt"... OK
Opening the root directory... OK
root directory:
    .
    ..
    numbers.txt
Closing the root directory... OK
Opening "/fs/numbers.txt"... lfs error:469: Corrupted dir pair at 3 2
Fail :(
error: unknown error (-52)
--- Mbed OS filesystem example ---
Mounting the filesystem... OK
Opening "/fs/numbers.txt"... OK
Incrementing numbers (10/10)... OK
Closing "/fs/numbers.txt"... OK
Opening the root directory... OK
root directory:
    .
    ..
    numbers.txt
Closing the root directory... OK
Opening "/fs/numbers.txt"... OK
numbers:
    0
    1
    10
    11
    12
    13
    14
    15
    16
    17
2
```

Added the following to main:

```c++
DataFlashBlockDevice bd(D11, D12, D13, D7);   
DigitalOut a(D10, 1);                         
DigitalOut b(D9, 1);                          
DigitalOut c(D8, 1);                          
DigitalOut d(D7, 1);                          
DigitalOut e(D6, 1);                          
DigitalOut f(D5, 1);                          
DigitalOut g(D4, 1);                          
```

geky, 2018-03-07:
Related https://github.com/ARMmbed/mbed-os-example-filesystem/issues/20

kegilbert, 2018-03-08:
I'll retry with the passthrough setup to avoid the pin conflicts.

adbridge, 2018-03-09:
@kegilbert any update on this ?

kegilbert, 2018-03-09:
@adbridge @geky 
It was the passthrough issue, closing when the docs changes go in.

closed 2018-03-09 by geky