ARMmbed/mbed-os-example-filesystem #37: Mutex error after pressing SW2 when using IAR build

SamiLappalainenArm, 2018-03-16:
build and run by using IAR 7.80.1.11873

after pressing SW2 button following error is shown in console output:
Mbed OS filesystem example done!
Mutex 20001df0 error -6: Not allowed in ISR context

when using GCC_ARM or ARM build then following output is shown after pressing SW2 button:
Mbed OS filesystem example done!
Initializing the block device... OK
Erasing the block device... OK
Deinitializing the block device... OK

SamiLappalainenArm, 2018-03-16:
same behaviour is seen if build is compiled with "--profile debug" option by using ARM

geky, 2018-03-17:
Heh, good catch. That erase function shouldn't be in irq context.

Will update with EventQueue next chance I get.

geky, 2018-03-17:
Actually that should be really easy to fix, @SamiLappalainenArm does https://github.com/ARMmbed/mbed-os-example-filesystem/pull/41 work for you?

SamiLappalainenArm, 2018-03-19:
problem solved by replacing `irq.fall(erase);` with `irq.fall(mbed_event_queue()->event(erase));`

closed 2018-04-05 by geky