ARMmbed/mbed-os-example-filesystem #39: differencies in output when using heap between GCC_ARM and ARM builds

SamiLappalainenArm, 2018-03-16:
used configuration:
// Physical block device
HeapBlockDevice bd(1024*512, 512);
 // File system declaration
LittleFileSystem fs("fs");

--------------------------------------------------------------------------------
* compiled by GCC_ARM
--------------------------------------------------------------------------------
Mounting the filesystem... lfs error:469: Corrupted dir pair at 0 1
lfs error:2202: Invalid superblock at 536873884 536874136
Fail :(
No filesystem found, formatting... OK
Opening "/fs/numbers.txt"... Fail :(
No file found, creating a new file... OK
Writing numbers (10/10)... OK
Seeking file... OK
Incrementing numbers (10/10)... OK
Closing "/fs/numbers.txt"... OK
Opening the root directory... OK
root directory:
    .
    ..
    numbers.txt
Closing the root directory... OK
Opening "/fs/numbers.txt"... OK
numbers:
    1
    2
    3
    4
    5
    6
    7
    8
    9
    10
Closing "/fs/numbers.txt"... OK
Unmounting... OK
Mbed OS filesystem example done!

--------------------------------------------------------------------------------
* compiled by ARM
--------------------------------------------------------------------------------
Mounting the filesystem... lfs error:469: Corrupted dir pair at 0 1
lfs error:2202: Invalid superblock at 1 536880404
Fail :(
No filesystem found, formatting... OK
Opening "/fs/numbers.txt"... Fail :(
No file found, creating a new file... OK
Writing numbers (10/10)... OK
Seeking file... OK
Incrementing numbers (10/10)... OK
Closing "/fs/numbers.txt"... OK
Opening the root directory... OK
root directory:
    .
    ..
    numbers.txt
Closing the root directory... OK
Opening "/fs/numbers.txt"... OK
numbers:
    1
    1
    2
    3
    4
    5
    6
    7
    8
    9
Closing "/fs/numbers.txt"... OK
Unmounting... OK
Mbed OS filesystem example done!

geky, 2018-03-17:
Hmm, this is precisely the issue that was fixed in https://github.com/ARMmbed/mbed-os-example-filesystem/pull/34

What hash of the example are you using?

SamiLappalainenArm, 2018-03-19:
>git log -n 1 mbed-os-5.8.0-oob
commit c76e4f121387fcff7cac233fbb0137c6e0bc6ba4 (HEAD -> mbed-os-5.8.0-oob, tag: mbed-os-5.8.0-rc2, origin/mbed-os-5.8.0-oob)

kegilbert, 2018-03-20:
The fix here:
29a91bf
is only on master it looks like, it needs to be cherry picked to the release branch.

@geky

0xc0170, 2018-06-08:
Is this issue still relevant? can be closed?

dannybenor, 2018-06-10:
Yes can be closed

closed 2018-06-10 by dannybenor