ARMmbed/mbed-os-example-filesystem #40: numbers stored on disk are not incremented after boot

SamiLappalainenArm, 2018-03-16:
following output has compiled by using ARM.

--- Mbed OS filesystem example ---
Mounting the filesystem... lfs error:469: Corrupted dir pair at 0 1
lfs error:2202: Invalid superblock at 64693 500
Fail :(
No filesystem found, formatting... OK
Opening "/fs/numbers.txt"... Fail :(
No file found, creating a new file... OK
Writing numbers (10/10)... OK
Seeking file... OK
Incrementing numbers (10/10)... OK
Closing "/fs/numbers.txt"... OK
Opening the root directory... OK
root directory:
    .
    ..
    numbers.txt
Closing the root directory... OK
Opening "/fs/numbers.txt"... OK
numbers:
    1
    1
    2
    3
    4
    5
    6
    7
    8
    9
Closing "/fs/numbers.txt"... OK
Unmounting... OK
Mbed OS filesystem example done!
--- Mbed OS filesystem example ---
Mounting the filesystem... OK
Opening "/fs/numbers.txt"... OK
Incrementing numbers (10/10)... OK
Closing "/fs/numbers.txt"... OK
Opening the root directory... OK
root directory:
    .
    ..
    numbers.txt
Closing the root directory... OK
Opening "/fs/numbers.txt"... OK
numbers:
    2
    1
    2
    3
    4
    5
    6
    7
    8
    9
Closing "/fs/numbers.txt"... OK
Unmounting... OK
Mbed OS filesystem example done!

SamiLappalainenArm, 2018-03-16:
compiling with GCC_ARM gives following output (seems to work correctly)

--- Mbed OS filesystem example ---
Mounting the filesystem... OK
Opening "/fs/numbers.txt"... OK
Incrementing numbers (10/10)... OK
Closing "/fs/numbers.txt"... OK
Opening the root directory... OK
root directory:
    .
    ..
    numbers.txt
Closing the root directory... OK
Opening "/fs/numbers.txt"... OK
numbers:
    5
    3
    4
    5
    6
    7
    8
    9
    10
    11
Closing "/fs/numbers.txt"... OK
Unmounting... OK
Mbed OS filesystem example done!
--- Mbed OS filesystem example ---
Mounting the filesystem... OK
Opening "/fs/numbers.txt"... OK
Incrementing numbers (10/10)... OK
Closing "/fs/numbers.txt"... OK
Opening the root directory... OK
root directory:
    .
    ..
    numbers.txt
Closing the root directory... OK
Opening "/fs/numbers.txt"... OK
numbers:
    6
    4
    5
    6
    7
    8
    9
    10
    11
    12
Closing "/fs/numbers.txt"... OK
Unmounting... OK
Mbed OS filesystem example done!

SamiLappalainenArm, 2018-03-16:
it seems that used filesystem does not have any impact to functionality only used complier has impact to the output

geky, 2018-03-17:
Also which block device are you using?

SamiLappalainenArm, 2018-03-19:
I have used block device definition shown below:

SDBlockDevice bd(
        MBED_CONF_SD_SPI_MOSI,
        MBED_CONF_SD_SPI_MISO,
        MBED_CONF_SD_SPI_CLK,
        MBED_CONF_SD_SPI_CS);

SamiLappalainenArm, 2018-03-19:
by pressing Button1 (SW2) before reset does not help:

--- Mbed OS filesystem example ---
Mounting the filesystem... Fail :(
No filesystem found, formatting... OK
Opening "/fs/numbers.txt"... Fail :(
No file found, creating a new file... OK
Writing numbers (10/10)... OK
Seeking file... OK
Incrementing numbers (10/10)... OK
Closing "/fs/numbers.txt"... OK
Opening the root directory... OK
root directory:
    numbers.txt
Closing the root directory... OK
Opening "/fs/numbers.txt"... OK
numbers:
    1
    1
    2
    3
    4
    5
    6
    7
    8
    9
Closing "/fs/numbers.txt"... OK
Unmounting... OK
Mbed OS filesystem example done!
Initializing the block device... OK
Erasing the block device... OK
Deinitializing the block device... OK
--- Mbed OS filesystem example ---
Mounting the filesystem... OK
Opening "/fs/numbers.txt"... OK
Incrementing numbers (10/10)... OK
Closing "/fs/numbers.txt"... OK
Opening the root directory... OK
root directory:
    numbers.txt
Closing the root directory... OK
Opening "/fs/numbers.txt"... OK
numbers:
    2
    1
    2
    3
    4
    5
    6
    7
    8
    9
Closing "/fs/numbers.txt"... OK
Unmounting... OK
Mbed OS filesystem example done!

kegilbert, 2018-03-19:
What's your Mbed OS and Filesystem example commit hash? We can try to debug locally.

geky, 2018-03-19:
@kegilbert I believe it's the same as https://github.com/ARMmbed/mbed-os-example-filesystem/issues/39

SamiLappalainenArm, 2018-03-20:
above comment is correct related to used commit hash.

kegilbert, 2018-03-20:
The fix here:
29a91bf
is only on master it looks like, it needs to be cherry picked to the oob branch.

@geky

SamiLappalainenArm, 2018-03-21:
this problem was solved after applying above fix

cmonr, 2018-06-11:
Closing since this is resolved.

closed 2018-06-11 by cmonr