ARMmbed/mbed-os-example-filesystem #48: K64F + SPIF + LittleFS using GCC_ARM

zvikas, 2018-06-03:
Setup: K64F + shield
I didn't modify any code or configuration (using default littlefs with spif)
Got error on boot:
mbed assertation failed: (int)obj->spi.instance != NC, file: .\mbed-os\targets\TARGET_Freescale\TARGET_MCUXpresso_MCUS\TARGET_M”¿“/ÿÿÿÿ
eÐ~

zvikas, 2018-06-03:
wrong issue

zvikas, 2018-06-03:
Using K64F with CI Shield 
After modified mbed_lib.json of spif_driver with the following changes:
      "K64F": {
             "SPI_MOSI": "D11",
             "SPI_MISO": "D12",
             "SPI_CLK":  "D13",
             "SPI_CS":   "D10"
        },

I got error :
--- Mbed OS file--- Mbed OS filesystem example ---

Mounting the filesystem... Fail :(

No filesystem found, formatting... Fail :(

error:  (-4001)

zvikas, 2018-06-04:
I was using wrong CI shield, going to try again with new shield.

closed 2018-06-04 by zvikas

geky, 2018-06-04:
Ah glad you were able to figure out the problem :+1: