ARMmbed/mbed-os-example-filesystem #62: Compilation stops before entering build phase

atomichan, 2018-09-07:
Hi.
During the OOB campaign, I found that mbed compile with argument -t IAR (--toolchain IAR) stops like the following:

`C:\work\OOB5.10\mbed-os-example-filesystem>mbed compile --toolchain IAR --target nucleo_f429zi
    [mbed] WARNING: If you're using Python 3 with Mbed OS 5.8 and earlier versions, Python errors will occur when compiling,
    testing and exporting
    ---
    [mbed] Auto-installing missing Python modules...    // freezes here.`

By the way, if I choose to use GCC_ARM, it works.

Could you help me what to do for problem determination?

Here are my environment information:
`
C:\work\OOB5.10\mbed-os-example-filesystem>python --version
Python 3.7.0
C:\work\OOB5.10\mbed-os-example-filesystem>pip --version
pip 18.0 from c:\users\atstom01\appdata\local\programs\python\python37-32\lib\site-packages\pip (python 3.7)
C:\work\OOB5.10\mbed-os-example-filesystem>mbed --version
1.8.0
`
The problem also occurred with Python 3.6.x and pip 10.x.

Thank you for your support.

Best regards,
Atsushi

ashok-rao, 2018-09-07:
Hi @atomichan , does this freeze for a long time? can you try with "-v" and paste the output here? Thanks.

atomichan, 2018-09-07:
Hi @ashok-rao , thank you for the followup.
The output of the trial in addition to the attribute "-v" is as follows.
The output about 30 min. after issuing the command).

`
C:\work\OOB5.10\mbed-os-example-filesystem>mbed compile -v --toolchain IAR --target nucleo_f429zi
[mbed] WARNING: If you're using Python 3 with Mbed OS 5.8 and earlier versions, Python errors will occur when compiling,
 testing and exporting

[mbed] Working path "C:\work\OOB5.10\mbed-os-example-filesystem" (program)
[mbed] Auto-installing missing Python modules...
[mbed] Exec "c:\users\atstom01\appdata\local\programs\python\python37-32\python.exe -u C:\work\OOB5.10\mbed-os-example-f
ilesystem\mbed-os\tools\make.py -t IAR -m nucleo_f429zi --source . --build .\BUILD\NUCLEO_F429ZI\IAR -v" in "C:\work\OOB
5.10\mbed-os-example-filesystem"
`

atomichan, 2018-09-07:
Hi @cmonr , here's the output of mbed ls:

mbed-os-example-filesystem (#9aa08635d7e5)
`- mbed-os (#3fb5781af180, tag: mbed-os-5.10.0-rc1)

In the same machine (which is a window 7 virtual machine on VMware Fusion), It looks like I can run IAR Embedded Workbench for ARM Version 7.80 and I can see a network license in IAR License Manager.
I use the same environment in OOB campaign 5.9.

cmonr, 2018-09-07:
@atomichan If you could humor me, can you find and run the iccarm.exe executable from the terminal?

For reference, mine was located here: `C:\Program Files (x86)\IAR Systems\Embedded Workbench 7.5\arm\bin`

atomichan, 2018-09-10:
@cmonr , @ashok-rao 
After the retrial, compilation was successfully finished without any additional configuration change.
I found only difference and it is an appearance of a dialogue box in starting IAR's license manager which say something is updated in the license.
Anyway I appreciate your support!

closed 2018-09-10 by atomichan