ARMmbed/mbed-os-example-filesystem #64: [OOB 5.10] Missing configuration item in API reference doc

coisme, 2018-09-07:
In the [API reference]( https://os-doc-builder.test.mbed.com/docs/development/reference/configuration-storage.html), an configuration item `nvstore.enabled` is missing.

My output from `mbed compile --config -v`:
```
Name: nvstore.area_1_address
    Description: Area 1 address
    Defined by: library:nvstore
    No value set
Name: nvstore.area_1_size
    Description: Area 1 size
    Defined by: library:nvstore
    No value set
Name: nvstore.area_2_address
    Description: Area 2 address
    Defined by: library:nvstore
    No value set
Name: nvstore.area_2_size
    Description: Area 2 size
    Defined by: library:nvstore
    No value set
Name: nvstore.enabled
    Description: Enabled
    Defined by: library:nvstore
    Macro name: NVSTORE_ENABLED
    Value: 1 (set by library:nvstore)
Name: nvstore.max_keys
    Description: Maximal number of allowed NVStore keys
    Defined by: library:nvstore
    Macro name: NVSTORE_MAX_KEYS
    Value: 16 (set by library:nvstore)
```

ashok-rao, 2018-09-11:
@geky .. can you please check this? Thanks.

kegilbert, 2018-09-11:
Good catch! Have a PR up at:

https://github.com/ARMmbed/mbed-os-5-docs/pull/724