ARMmbed/mbed-os-example-filesystem #81: Readme needs update

jafermarq, 2019-01-15:
I believe the Readme, in particular, the `changing-the-block-device` section, needs to be updated after the latest change to `main.cpp`. Also, shouldn't the SPI numbering be specified in the `mbed_app.json`. Although I'm able to compile this project, executing the generated binary will result in a hard fault.

dannybenor, 2019-01-17:
@jafermarq can you provide more specific info? What target are you using? Did you change anything?

jafermarq, 2019-01-18:
I'm using a Nucleo-F767ZI and `mbed-cli 1.8.3`. I can compile the example in `master` but when flashing it, it will result in a HardFault when attempting to mount the filesystem. This error seems to trigger a reset so this is happening in a loop over and over. The error code is `0x800259E` By switching to `override-storage-components` branch I can make it work. I'm mostly interested in using a SD card, so I haven't tested other types of `BlockDevice` interfaces.

dannybenor, 2019-01-20:
@jafermarq Did you add support for Nucleo-F767ZI in targets.json? Can you paste here your compilation command?

jafermarq, 2019-01-20:
@dannybenor, I haven't edited `targets.json`. I run the following command: `mbed compile -m auto -t GCC_ARM`.

RonEld, 2019-06-04:
Hi @jafermarq 
What is set as you component in your target?
Did you add
~~~
"target_overrides": {
         ...
         "NUCLEO_F767ZI": {
             "target.components_add": ["SD"],
             ...
         },
         ...
  }
~~~
to your `mbed_app.json` ?

linlingao, 2019-07-22:
@jafermarq Can you respond to @RonEld? If you've solved your issue, please close.

adbridge, 2019-11-01:
Apologies, this was closed in error by the automated bot - re-opening.

evedon, 2020-12-24:
Fixed by #133

PR #133 (2020-12-23): Improve description of proces to add a new block device
Fixes #81

Updated README to reflect the current procedure for adding a block device. Replaced extraneous information regarding types of block devices supported by Mbed OS with link to corresponding page on Mbed OS documentation website. Fixed mbedtools invocation in travis script which was breaking CI.
closed 2020-12-24 by evedon