ARMmbed/mbed-os-example-filesystem #88: No mention of using NAND flash block devices.

peter-tatch, 2019-02-21:
I would like to see some explanation of whether this can work with NAND flash. Can any of the block devices work with raw serial NAND flash? Can we configure the SPIFBlockDevice to work with NAND flash for example. If not, what is the limitation that prevents a NAND block device from being included?

peter-tatch, 2019-02-21:
Thank you but that site appears to be down. Is there a new link or is it temporarily down?

cmonr, 2019-02-21:
@peter-tatch Sorry, but the link is internal-only. It's our way of syncing multiple sources of issue reporting into one collected bucket.

dannybenor, 2019-03-05:
@peter-tatch NAND is a different technology, that requires the management of bad sectors. It is used in SD cards, therefore the possible way to use NAND in Mbed OS right now is via SD block device

adbridge, 2019-11-01:
Apologies, this was closed in error by the automated bot - re-opening.

evedon, 2020-11-27:
See further example of changing the block device in https://github.com/ARMmbed/mbed-os-example-blockdevice#changing-the-block-device.

Closing since it seems like this question was answered.

closed 2020-11-27 by evedon