ARMmbed/mbed-os-example-filesystem #98: No free space found in LFS when mounted to FlashIAPBlockDevice

DrynnBavis, 2019-05-16:
I'm running this example on a UBLOX_EVK_ODIN_W2 board and can't seem to seek or save the written to file. fseeking the file after writing to it causes this warning: 

lfs warn:314: No more free space 4

and then when closing it I finally receive a fatal run-time error:

error: No space left on device (-28)

Is there a way to check how much space is available in this LFS? The FlashIAP Block Device I mounted it to has been given 512 KB of space. Is it possibly not using any of this? Running '>> du <my-device-name>' returns 0 B.

davidsaada, 2019-05-16:
@geky

DrynnBavis, 2019-05-16:
Sorry, you can close this. I found the solution in your README mentioning LFS requires at least 6*bd.erase_size() of memory allocation. My erase size is 131k so this is huge for me, but at least it's working now. Error was definitely throwing me off though. Could it be changed to mention something about "Insufficient block device size detected, minimum of 6 blocks required for LFS".

geky, 2019-05-16:
Ah yes, LittleFS v1 does not work well with few blocks.

It's experimental, but v2 of LittleFS removes this limitation.

You can try it out by adding this library and changing any instances of LittleFileSystem to LittleFileSystem2:
https://github.com/armmbed/mbed-littlefs

davidsaada, 2019-05-16:
Thanks for that.

closed 2019-05-16 by davidsaada

DrynnBavis, 2019-05-16:
> Ah yes, LittleFS v1 does not work well with few blocks.
> 
> It's currently experimental, but v2 of LittleFS removes this limitation.
> 
> You can try it out by adding this library and changing any instances of LittleFileSystem to LittleFileSystem2:
> https://github.com/armmbed/mbed-littlefs

The current dev board I'm on has a much larger block size than the on we'll be using for prod so it's no worry down the road. But thanks for mentioning the second version, might end up using it down the road!

DBS06, 2019-05-28:
@DrynnBavis 

> I found the solution in your README mentioning LFS requires at least 6*bd.erase_size() of memory allocation.

In which README is this mentioned?