ARMmbed/mbed-os-example-filesystem #141: CMake build fails with `Unknown CMake command "mbed_set_mbed_target_linker_script"`

Patater, 2021-04-08:
### Description of defect

```
#### STDERR ####

#### End of STDOUT/STDERR ####

IMAGE is /builds/workspace/mbed-os-ci_cmake-example-GCC_ARM-dev@11/examples/mbed-os-example-devicekey/cmake_build/DISCO_L475VG_IOT01A/develop/GCC_ARM/mbed-os-example-devicekey.bin
[EXAMPLES]> INFO     In folder 'mbed-os-example-thread-statistics'
[EXAMPLES]> INFO     In folder 'mbed-os-example-sys-info'
[EXAMPLES]> INFO     In folder 'mbed-os-example-cpu-stats'
[EXAMPLES]> INFO     In folder 'mbed-os-example-socket-stats'
[EXAMPLES]> INFO     In folder 'mbed-os-example-error-handling'
[EXAMPLES]> INFO     In folder 'mbed-os-example-filesystem'
[EXAMPLES]> INFO     Compiling mbed-os-example-filesystem DISCO_L475VG_IOT01A GCC_ARM
[EXAMPLES]> INFO     Executing command 'mbed-tools compile -t GCC_ARM -m DISCO_L475VG_IOT01A -c'...

#### STDOUT ####
Configuring project and generating build system...
-- The C compiler identification is GNU 9.2.1
-- The CXX compiler identification is GNU 9.2.1
-- The ASM compiler identification is GNU
-- Found assembler: /usr/local/gcc-arm-none-eabi-9-2019-q4-major/bin/arm-none-eabi-gcc
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/local/gcc-arm-none-eabi-9-2019-q4-major/bin/arm-none-eabi-gcc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/local/gcc-arm-none-eabi-9-2019-q4-major/bin/arm-none-eabi-g++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Found Python3: /usr/bin/python3.7 (found version "3.7.4") found components: Interpreter 
-- Checking for Python package prettytable -- found
-- Checking for Python package future -- found
-- Checking for Python package jinja2 -- found
-- Checking for Python package intelhex -- found
-- Configuring incomplete, errors occurred!
See also "/builds/workspace/mbed-os-ci_cmake-example-GCC_ARM-dev@11/examples/mbed-os-example-filesystem/cmake_build/DISCO_L475VG_IOT01A/develop/GCC_ARM/CMakeFiles/CMakeOutput.log".
See also "/builds/workspace/mbed-os-ci_cmake-example-GCC_ARM-dev@11/examples/mbed-os-example-filesystem/cmake_build/DISCO_L475VG_IOT01A/develop/GCC_ARM/CMakeFiles/CMakeError.log".

#### STDERR ####
CMake Error at CMakeLists.txt:18 (mbed_set_mbed_target_linker_script):
  Unknown CMake command "mbed_set_mbed_target_linker_script".
```

#### Target(s) affected by this defect ?
DISCO_L475VG_IOT01A

#### Toolchain(s) (name and version) displaying this defect ?
GCC_ARM

#### What version of Mbed-os are you using (tag or sha) ?

9bbc8c914c7e 

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)
Mbed CLI 2

#### How is this defect reproduced ?
`mbed-tools compile -t GCC_ARM -m DISCO_L475VG_IOT01A -c`

LDong-Arm, 2021-04-08:
PR created: https://github.com/ARMmbed/mbed-os-example-filesystem/pull/142

LDong-Arm, 2021-04-13:
Fix merged.

closed 2021-04-13 by LDong-Arm

0xc0170, 2021-04-13:
Fixed via #141 (it was not automatically closed?).  I was about to close it @LDong-Arm  was faster :)

PR #142 (2021-04-13): Fix build with Mbed CLI 2 and update README.md
`CMakeLists.txt`:
* `mbed_set_mbed_target_linker_script()` is no longer present in Mbed OS
* `mbed_configure_app_target()` is not required to be called by the application's `CMakeLists.txt`

`README.md`:
* Add instruction for Mbed CLI 2

Fixes #141
commit 0cec6f9b (2021-04-13): Fix Mbed CLI 2 support

`CMakeLists.txt`:
* mbed_set_mbed_target_linker_script() is no longer present in
  Mbed OS
* mbed_configure_app_target() is not required to be called by the
  application's `CMakeLists.txt`

Fixes #141